OmniLRS/OmniLRS #16: Lunalab Prim Error

KhaledSharif, 2024-05-10:
```bash
cd OmniLRS  # commit fe1d57
alias PYTHON_PATH=~/.local/share/ov/pkg/isaac_sim-2023.1.1/python.sh
PYTHON_PATH run.py environment=lunalab mode=ROS2 rendering=ray_tracing
```

```
Error executing job with overrides: ['environment=lunalab', 'mode=ROS2', 'rendering=ray_tracing']
Traceback (most recent call last):
  File "/home/khaled/Documents/GitHub/OmniLRS/run.py", line 73, in run
    SM, simulation_app = startSim(cfg)
  File "/home/khaled/Documents/GitHub/OmniLRS/src/environments_wrappers/__init__.py", line 32, in startSim
    SM = ROS2_SimulationManager(cfg, simulation_app)
  File "/home/khaled/Documents/GitHub/OmniLRS/src/environments_wrappers/ros2/simulation_manager_ros2.py", line 118, in __init__
    self.ROSRobotManager.RM.preloadRobot(self.world)
  File "/home/khaled/Documents/GitHub/OmniLRS/src/robots/robot.py", line 75, in preloadRobot
    self.addRRG(
  File "/home/khaled/Documents/GitHub/OmniLRS/src/robots/robot.py", line 134, in addRRG
    rrg.initialize(world)
  File "/home/khaled/Documents/GitHub/OmniLRS/src/robots/robot.py", line 335, in initialize
    rigid_prim = RigidPrim(
  File "/home/khaled/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/prims/rigid_prim.py", line 115, in __init__
    self._rigid_prim_view = RigidPrimView(
  File "/home/khaled/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/prims/rigid_prim_view.py", line 144, in __init__
    XFormPrimView.__init__(
  File "/home/khaled/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/prims/xform_prim_view.py", line 126, in __init__
    raise Exception(
Exception: Prim path expression /Robots/ex1/left_front_wheel_link is invalid, a prim matching the expression needs to created before wrapping it as view
```

AntoineRichard, 2024-05-10:
@jnskkmhr is it possible that you updated the USD of the EX1 robot? That could explain why it’s not finding a matching prim in the regex.

jnskkmhr, 2024-05-12:
@KhaledSharif 
Hi, thank you for reporting the error. 
I have uploaded the robot USD file and updated the Gdrive link in README. 
So, you should see errors resolved in the latest main branch. 

As for now, I only tested the robot teleoperation in ROS1, and I have not added ROS2 Omnigraph. 
Shortly after ICRA, we will release the sets of demonstrations in ROS1 and ROS2 to the main branch.

closed 2024-05-12 by jnskkmhr