OmniLRS/OmniLRS #23: Weird jolts during drives

TL-4319, 2024-08-16:
First and foremost, thank you for the great work. I have been experimenting with the simulator and I seeing some behavior during driving where sometime there are just random jolts that shake the rover. The video had modification made to the physics scene where gravity is set to 1.62 m/s^2 but the behavior still exist at the default gravity of 9.81. Changing the simulation dt to 0.01 doesn't seem to get rid of it completely.

[Screencast from 08-16-2024 02:58:56 PM.webm](https://github.com/user-attachments/assets/54335c61-5e9f-4135-9573-3d7826287de5)

I would like to ask if you guys are also seeing this phenomenon during your runs and have some suggestion of what could be done? I am currently reading into different collision mesh by isaac sim as well.

Thank you

AntoineRichard, 2024-08-20:
Hey there, thanks for the feedback! We are aware of the problem.

I think this is related to the way the collider mesh is generated, though I'm not 100% certain.

I am currently busy with other projects, but I was planning on investigating this in early September.
I'll update this issue when the fix is release.

Cheers,

Antoine

TL-4319, 2024-08-21:
A fix that I found recently is to change the physics scene collision system from PCM to SAT

This is the lunaryard 20m environment with PCM collision
[Screencast from 08-19-2024 09_56_52 AM.webm](https://github.com/user-attachments/assets/281fc8d2-7198-4b3f-97fc-54a2bdad343e)

This is the lunaryard 20m environment with SAT collision
[Screencast from 08-19-2024 10_12_27 AM.webm](https://github.com/user-attachments/assets/2c4a0ad9-d368-4b17-b425-9ab92c9d7541)

Right now, the change from PCM to SAT is done via Isaac Sim GUI. I have not found a way to do it programatically yet even though I check several doc for physicx and Isaac sim so I don't really have code changes for a PR for now. Hopefully, something can be found regarding that with more research.  

Even then, during drives, i need to mess around with the wheel collider material by adding some compliance and dampening. So far it has been trying different things randomly but I plan on doing more experimentation over the weekends. I'll share the finding later if there are any.

AntoineRichard, 2024-08-21:
Did you try on Isaac 4.1.0 ?

TL-4319, 2024-08-21:
This is run on Isaac Sim 2023.1.1 following the instruction in the readme. I'm not sure either if they have tested much with uneven terrain mesh either based on what I have searched on their forum.
Is the coming release migrating to Isaac 4.1.0?

AntoineRichard, 2024-09-03:
Just to confirm that this is an issue in Isaac 2023.1.1, it's no longer happening in 4.1.0. We did some tests, and should be able to switch to 4.1.0 if need be. I think I also have a fix for Isaac 2023.1.1. We will release the update together with new envs in a week or so.

AntoineRichard, 2024-09-09:
@TL-4319 this should be fixed now? The new version of the stack should be more robust. The doc will be rolling in the next 2 days.

TL-4319, 2024-09-10:
That's great to hear! So Isaac 4.1.0 should fix this problem? I saw on the readme of OmniLRS V2 that you are encountering some weird rendering problem? Can you briefly explain that so I can decide on if switching to 4.1.0 is the best choice? Thanks

AntoineRichard, 2024-09-10:
It's fixed in 2023.1.1 as well. Or at least I don't have the issue **too much**.

For the bug you check it out here.
https://docs.google.com/document/d/1ZgLjNSMZJYe_n1JQaif4JvXOjEgFlmpbwR7uobstjvk/edit

If you try it, let me know.

TL-4319, 2024-09-14:
I performed testing with physics with OmniLRS v2. I was not able to test it with the provided Husky and Jackal USD due to a crash issue mentioned in issue #25 so I did the test with the viper.usd.

The physics did improve for the Lunaryard environment. However, it is worse with the largescale environment. Changing collision system to SAT improved it slightly but the problem persists. 

I will perform some more test with wheel collider material tuning to see if it fixes. If the problem persist, I will try IsaacSim 4.1.0

Some videos of the behaviors will come as I perform more tests

OldCatBruce, 2024-10-31:
May I ask how you make the vehicle move？Thank you!

TL-4319, 2024-10-31:
I was using a joystick and ROS2 joy_node to control the robot. You'll need to make sure that the robot you are using has an Action Graph that subscribes to the twist command from ROS2.

OldCatBruce, 2024-10-31:
Thank you for your answer. I will try again before communicating with you.

TTXDY, 2025-01-03:
> First and foremost, thank you for the great work. I have been experimenting with the simulator and I seeing some behavior during driving where sometime there are just random jolts that shake the [...]

Hello, hello. After I ran it, I found that the terrain didn't deform and I didn't see the wheel tracks. Could you please tell me what I should do? I set the physical properties of the wheels as rigid bodies and the ground as deformable.

ljburtz, 2026-03-20:
thank you for sharing the video and explanation. there is something unexpected indeed (even with the husky actually).

one thing to look at is how the collider mesh is defined for your spoke wheels. you can try different options 
`Convex hull`, `Convex decomposition` , or `SDF`

if you are able to share the usd of your robot i can try to replicate on my side.

busraburhan, 2026-03-23:
I’m not sure whether I’m allowed to share the model of my robot, since I’m just an intern here :) However, the colliders are defined the same as the mesh—they are not simplified shapes like cylinders or boxes. I also think that if the colliders were incorrect, the issue would appear in the LunaLab environment as well. Do you have any idea why collider-related problems would only occur at large scales? @ljburtz