OmniLRS/OmniLRS #32: Unable to launch any simulation environments using the docker install

TedVanderfeen, 2024-10-08:
Hey guys,

I've followed the instructions for the docker install to get OmniLRS running but it doesn't seem to be configured correctly. I am able to successfully install the docker image and pull all the assets down but when I attempt to run even the default simulation environment using "/isaac-sim/python.sh run.py" it crashes out. Have attached my log file, notable errors are the "ModuleNotFoundError" for the empy package and everything from line 369-379. The simulation program runs until successfully loading the simulation app and mesh, where it then hangs for approx 2 minutes and then crashes with the final 10 lines of errors.

I have tried a few things such as installing the packages it says are missing, which are usually already installed when I run "/isaac-sim/python.sh -m pip install xxxxx" and adding /isaac-sim/kit/python/ to $PATH but as of yet have not been able to get this working. Would appreciate any and all assistance you can provide.

AntoineRichard, 2024-10-08:
Hey! Thanks for trying OmniLRS, I'm sorry this is not working for you :/.
I think this is a nucleus issue, but I'm not certain. The missing module is fine though.

Could you run the following command, and before you do create the file as detailed below?
`/isaac-sim/python.sh run.py environment=lunalab_no_robot`

To test this, you need to create a new file inside `cfg/environment` named `lunalab_no_robot.yaml`.
Inside this file, please put the following. Let me know how it goes. If it does it's that the robot still has some of its assets in the cloud.

```yaml
name: Lunalab
seed: 42  
physics_dt: 0.0333 # 30 Hz
rendering_dt: 0.0333 # 30 Hz
enforce_realtime: true

lunalab_settings: # These settings cannot be changed.
  lab_length: 10.0
  lab_width: 6.5
  resolution: 0.01

# Rocks sampling parameters.
rocks_settings:
  instancers_path: /Lunalab/Rocks
  rocks_settings:
    large_rocks: # The name doesn't matter.
      seed: ${....seed}
      collections: ["lunalab_rocks"] # Where to get the rock models from.
      use_point_instancer: True # If True, the rocks will be instanced using the PointInstancer.
                                # If False, it will use the custom instancer that works for SDG.
      requests: # A list of request used to distribute the rocks.
        req_pos_xy: 
          attribute: Position
          axes: ["x", "y"]
          layer:
            name: Image
            # data: Is loaded automatically from the DEM.
            mpp_resolution: ${.......lunalab_settings.resolution}
            output_space: 2
          sampler:
            name: HardCoreUniform
            randomization_space: 2
            seed: ${.......seed}
            core_radius: 0.5
            num_repeat: 2
            min: ${as_tuple:0.5,0.5}
            max: ${as_tuple:6,9.5}

        req_pos_z:
          attribute: Position
          axes: ["z"]
          layer:
            name: Image
            output_space: 1
          sampler:
            name: Image
            randomization_space: 1
            # resolution: Resolution is infered automatically from the loaded DEM.
            # data: Is loaded automatically from the DEM.
            mpp_resolution: ${.......lunalab_settings.resolution}

        req_random_z_rot:
          attribute: Orientation
          axes: ["x", "y", "z", "w"]
          layer:
            name: RollPitchYaw
            rmax: 0
            rmin: 0
            pmax: 0
            pmin: 0
            ymax: 6.28318530718
            ymin: 0
          sampler:
            name: Uniform
            randomization_space: 3
            seed: ${.......seed}

        req_scale:
          attribute: Scale
          axes: ["xyz"]
          layer:
            name: Line
            xmin: 1.0
            xmax: 1.0
          sampler:
            name: Uniform
            randomization_space: 1
            seed: ${.......seed}

terrain_manager:
  moon_yard:
    crater_generator:
      profiles_path: assets/Terrains/crater_spline_profiles.pkl
      min_xy_ratio: 0.85
      max_xy_ratio: 1.0
      resolution: ${....lunalab_settings.resolution}
      pad_size: 500
      random_rotation: True
      z_scale: 1.0
      seed: ${....seed}

    crater_distribution:
      x_size: 10.0
      y_size: 6.5
      densities: [0.025, 0.05, 0.5]
      radius: [[1.5,2.5],[0.75,1.5],[0.25,0.5]]
      num_repeat: 0
      seed: ${....seed}

    base_terrain_generator:
      x_size: ${....lunalab_settings.lab_length}
      y_size: ${....lunalab_settings.lab_width}
      resolution: ${....lunalab_settings.resolution}
      max_elevation: 0.25
      min_elevation: -0.025
      z_scale: 1.0
      seed: ${....seed}
    
    deformation_engine:
      enable: False
      delay: 2.0
      terrain_width: ${....lunalab_settings.lab_width}
      terrain_height: ${....lunalab_settings.lab_length}
      terrain_resolution: ${....lunalab_settings.resolution}
      footprint:
        width: 0.09
        height: 0.2
      deform_constrain:
        x_deform_offset: 0.0
        y_deform_offset: 0.0
        deform_decay_ratio: 0.01
      boundary_distribution: 
        distribution: trapezoidal
        angle_of_repose: 1.047 #pi/3
      depth_distribution: 
        distribution: trapezoidal
        wave_frequency: 4.14 # num_grouser/pi (no slip)
      force_depth_regression:
        amplitude_slope: 0.00006
        amplitude_intercept: 0.008
        mean_slope: -0.00046
        mean_intercept: -0.0013

    is_yard: False
    is_lab: True

  root_path: /Lunalab
  texture_path: /Lunalab/Looks/Basalt
  dems_path: Terrains/Lunalab
  mesh_position: [0, 0, 0]
  mesh_orientation: [0, 0, 0, 1] # Quaternion x,y,z,w
  mesh_scale: [1, 1, 1]
  sim_length: ${..lunalab_settings.lab_length}
  sim_width: ${..lunalab_settings.lab_width}
  resolution: ${..lunalab_settings.resolution}

robots_settings:
  uses_nucleus: False
  is_ROS2: True
  max_robots: 5
  robots_root: "/Robots"
  parameters:
```

TedVanderfeen, 2024-10-08:
That was indeed the problem! Thanks, Antoine. I just had a look at the page that is linked in the setup guide for installing the Isaac sim docker prerequisites and it doesn't mention anything about installing Nucleus until after the requirements steps. So I had assumed that the cloud version of nucleus would be sufficient. I'll install Nucleus server now and report back.

AntoineRichard, 2024-10-08:
I think it'be better if we could completely remove the dependencies to Nvidia's remote cloud assets. Let me know if it works, and in the meantime I'll see if I can modify the assets.
I'll also move the Nucleus requirements higher up so that it's clearer that Nucleus is needed.
Thanks for reporting this.

TedVanderfeen, 2024-10-08:
I have installed Nucleaus server on the docker host and it looks like I can run the lunalab and the lunaryard envs now, but the largescale sim is still crashing (logs attached). Good idea on moving away from Nucleus dependent assets, It could be firewall rules at my organization that are blocking Nucleus cloud connections but I'm unsure.

Thanks again, Antoine.

[largescale_crash.log](https://github.com/user-attachments/files/17299968/largescale_crash.log)

AntoineRichard, 2024-10-09:
Hey Ted! I'm not quite sure why, I don't see any errors popping up. The map seems to be generated just fine, but something is triggering the shutdown of the workers (and it's either a crash from python or a ctrl+c). Is there something more in your terminal that is not reported in the logs? I did run a couple sanity checks on my machine with docker and I don't see any errors.

Could you check the following:
- 1. Is there some explicit error message in the terminal output.
- 2. Are you loading a robot?
- 3. Did the DEMs from NASA get downloaded?

Ideally having the terminal output with the errors would be great, you can just copy and paste everything after Isaac loaded.
Let me know!

TedVanderfeen, 2024-10-09:
My apologies, didn't realise that all the useful error information got cut off. Have included a new log file directly copied from the terminal window.

- i. The first thing that I can see going wrong in the terminal is "Error executing job with overrides: ['environment=largescale']" which seems to point to a problem with the largescale.yaml file. But I haven't modified it since cloning the repo.
- ii. Yes, loading the husky as originally defined in largescale.yaml
- iii. DEMs did indeed get downloaded, my asset folder structure matches that of the installation guide (assets/Terrain/SouthPole/(dems_folders_surf)/(dem.npy + dem.yaml).

AntoineRichard, 2024-10-10:
Oh Whoops. I need to start building unit tests....
That was an issue with numba not knowing what the logger was in "no python" mode.
I've just pushed a fix.

TedVanderfeen, 2024-10-14:
Confirming that the largescale environment is working as expected now. Thanks, Antoine.

closed 2024-10-14 by TedVanderfeen