OmniLRS/OmniLRS #33: Largescale DEM render artifacts at the edges

TL-4319, 2024-10-11:
Hi Antoine,

Hope all is going well. I have a quick question regarding a rendering artifact at the edge of the DEM when using the large scale environment. I attached the images.

The first image show the artifacts when zoomed out which seems to be some weird behavior at the edges of the downloaded DEM.
[image]

The second image shows that it does get rendered from an attached camera. I also see the artifact when viewing from RVIZ.
[image]

My quick fix for it now is to move the starting position of the robot to be further center of the loaded DEM so i don't even see the artifacts but I wonder if there is a way to fix it correctly.

Thanks in advance

AntoineRichard, 2024-10-15:
Hi there!

These are not rendering artifacts. That is the intended behavior. It is either that, or just a black background. If you'd rather not see it let me know. I can add a parameter in the environment config.

TL-4319, 2024-10-15:
I would like to not render those for our application. Please let me know when you have added the feature. Thanks

TL-4319, 2024-10-15:
I'm in no rush for that at the moment because of other projects at the same time. Thank you!