OmniLRS/OmniLRS #34: how to solve this error?

dzy666fly, 2024-10-15:
Thank u so much for sharing the code!
When I use bulid_docker.sh, it shows 
`docker build /OmniLRS-main/omnilrs.docker --tag isaac-sim-omnilrs
[+] Building 0.7s (2/2) FINISHED                                                                                                           docker:default
 => [internal] load build definition from Dockerfile                                                                                                 0.0s
 => => transferring dockerfile: 1.65kB                                                                                                               0.0s
 => ERROR [internal] load metadata for nvcr.io/nvidia/isaac-sim:2023.1.1                                                                             0.5s
------
 > [internal] load metadata for nvcr.io/nvidia/isaac-sim:2023.1.1:
------
Dockerfile:1
--------------------
   1 | >>> FROM nvcr.io/nvidia/isaac-sim:2023.1.1
   2 |     
   3 |     RUN /isaac-sim/python.sh -m pip install opencv-python omegaconf hydra-core
--------------------
ERROR: failed to solve: nvcr.io/nvidia/isaac-sim:2023.1.1: failed to resolve source metadata for nvcr.io/nvidia/isaac-sim:2023.1.1: unexpected status from HEAD request to https://nvcr.io/v2/nvidia/isaac-sim/manifests/2023.1.1: 401 Unauthorized`

However, if I purely use `docker pull nvcr.io/nvidia/isaac-sim:2023.1.1`, it works without errors.

So how can I solve this problem?

Thank u so much.

dzy666fly, 2024-10-15:
LOL. What I need is to pull isaac-sim first and then build the docker.

AntoineRichard, 2024-10-15:
Hi there!

Thanks for using OmniLRS!

This is not the desired behavior, it should be able to pull IsaacSim's image on its own.
I could not reproduce this issue when running: `omnilrs.docker/build.sh` (I have cleaned everything so that it was built from scratch)

Could you kindly provide the whole of the error trace?

Thanks,

Antoine

dzy666fly, 2024-10-15:
Thank u so much for reply. I have solved this problem as what I said.

dzy666fly, 2024-10-15:
Now u can close this issue. I cannot close it.

closed 2024-10-15 by AntoineRichard