OmniLRS/OmniLRS #35: ERROR when exacting DEMS

dzy666fly, 2024-10-15:
Thank u again for sharing this wonderful work.

When I use native install step by step, `./scripts/extract_dems_override.sh` goes wrong.

```Extracting DEMs...
Reading DEMs info...
ERROR 1: TIFFFetchDirectory:/home/zhenyuan/Code/Source/OmniLRS-main/tmp/ldem_87s_5mpp.tif: Can not read TIFF directory count
ERROR 1: TIFFReadDirectory:Failed to read directory at offset 2522680612
ERROR 1: TIFFFillTile:Read error at row 19456, col 19456, tile 10439; got 92479 bytes, expected 129039
ERROR 1: TIFFReadEncodedTile() failed.
ERROR 1: /home/zhenyuan/Code/Source/OmniLRS-main/tmp/ldem_87s_5mpp.tif, band 1: IReadBlock failed at X offset 77, Y offset 66: TIFFReadEncodedTile() failed.
Finished extracting DEMs.
Moving DEMs to /home/zhenyuan/Code/Source/OmniLRS-main/assets/Terrains/SouthPole...
Finished moving DEMs.
Cleaning up...
Finished cleanup.
```

So how can I solve it?

Thank u so much.

AntoineRichard, 2024-10-16:
Hum, that's coming from NASA so it should be fine, but sometimes the download fails. One way to do it is to manually download the map and redo the processing on it. You could try the following in the docker:
```bash
mkdir tmp
cd tmp
wget https://pgda.gsfc.nasa.gov/data/LOLA_5mpp/87S/ldem_87s_5mpp.tif
gdalinfo ldem_87s_5mpp.tif > ldem_87s_5mpp.info
cd ..
./scripts/extract_dems_override.sh
```

dzy666fly, 2024-10-17:
Thank u so much for reply so quickly. It solves, though I have rerun is the LunarLab which maybe doesnot use this DEM data.

closed 2024-10-17 by dzy666fly