OmniLRS/OmniLRS #37: References an asset that can not be found: '../../Textures/Earth/earth_color_with_clouds.tif

dzy666fly, 2024-10-17:
Hi, it's me again :rofl: 

After native installing, command `Softwares/omni/pkg/isaac-sim-4.1.0/python.sh run.py environment=lunaryard_20m` shows error as follows:
```
2024-10-17 09:34:40 [161,147ms] [Error] [omni.hydra] [UsdToMdl] Prim '/Lunaryard/Earth/Looks/OmniPBR/Shader' parameter 'diffuse_texture': References an asset that can not be found: '../../Textures/Earth/earth_color_with_clouds.tif'
```

Well, `earth_color_with_clouds.tif` can be found in the folder `OmniLRS-main/assets/Textures/Earth`. And I found that nothing in the folder `OmniLRS-main/assets/Terrains/Lunaryard`, maybe it is the reason or not?

How can I solve this error?

Thank u so much forever :smile:

AntoineRichard, 2024-10-17:
No worries, thanks for finding issues :)
Could you kindly send me a screenshot similar to this one, where I can see what your scene looks like?
[image]

Maybe I messed up with one of the assets.

dzy666fly, 2024-10-17:
Of course, and I even screenshot the error information in the picture.

[image]

dzy666fly, 2024-10-17:
I am sorry that the picture is not same as u mentioned because the network did not load the full picture. Plz allow me to reupload again now.

dzy666fly, 2024-10-17:
And I observe that even if I set up this term as in follows, nothing seems to change visually.

[image]

TL-4319, 2024-10-18:
This seems to only happen in isaacsim 4.1.0 as I don't have this issue in isaacsim 2023.1.1. @dzy666fly when you click on Earth prim and press "F", the camera should be taken there and you should see the correct texture on the Earth model. Again, the sim very far away from origin so it might not be easy to pick up.

[image]

I'm still not sure why IsaacSim can't find the texture in 4.1.0 but this error should not affect other aspect unless you need to actually track the Earth in your application

AntoineRichard, 2024-10-21:
Hey, I checked and could not reproduce this issue with Isaac 2023.1.1. I'll try to look into why this is happening with 4.1.0. I could be tied to the format, maybe they don't like this type of tiff in 4.1.0. I'll look into it.

dzy666fly, 2024-10-21:
> This seems to only happen in isaacsim 4.1.0 as I don't have this issue in isaacsim 2023.1.1. @dzy666fly when you click on Earth prim and press "F", the camera should be taken there and you should see the correct texture on the Earth model. Again, the sim very far away from origin so it might not be easy to pick up.
> 
> ![Screenshot from 2024-10-18 12-52-51](https://private-user-images.githubusercontent.com/66798302/377965197-32a98cc2-9b6e-439c-872e-42eb96984025.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0OTkwMzAsIm5iZiI6MTcyOTQ5ODczMCwicGF0aCI6Ii82Njc5ODMwMi8zNzc5NjUxOTctMzJhOThjYzItOWI2ZS00MzljLTg3MmUtNDJlYjk2OTg0MDI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIxVDA4MTg1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjZDAwMmVlMzI2NzY0YmFmMTMyMDdjZjVhYWNiMGZlN2E3NDEwNTJmYzBhZWNmYmJkZmFkYmI0MDkyOGVjN2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.KUMQ3t_MOOQWlqjtpwdPDFQ80nS-i7cmWIc23HB1bF8)
> 
> I'm still not sure why IsaacSim can't find the texture in 4.1.0 but this error should not affect other aspect unless you need to actually track the Earth in your application

Thank u so much

dzy666fly, 2024-10-21:
@TL-4319 Thank us so much, bro :laughing: 

After changing the folder from `../../Textures/Earth/earth_color_with_clouds.tif` to `OmniLRS-main/assets/Textures/Earth/earth_color_with_clouds.tif`, it works when I click on Earth prim and press "F"

closed 2024-10-21 by dzy666fly