OmniLRS/OmniLRS #38: cannot see the scene when running the LargeScale

dzy666fly, 2024-10-17:
Hi, it's me again :laughing: 

After native installing, when run `python.sh run.py environment=largescale`, nothing seemed to be wrong but the scene is blank as shown in the picture.

[image]

But the shell seems to be OK.
```
[2024-10-17 22:23:53,511][src.terrain_management.large_scale_terrain.high_resolution_DEM_generator][DEBUG] - collecting...
[2024-10-17 22:23:53,705][src.terrain_management.large_scale_terrain.high_resolution_DEM_generator][DEBUG] - Thread closing map is done
[2024-10-17 22:23:53,706][src.terrain_management.large_scale_terrain.map_manager][INFO] - HR DEM initialized in 104.29600381851196 seconds
terrain updated
coordinates aquired
Module src.terrain_management.large_scale_terrain.geometry_clipmaps_warp 2e7d857 load on device 'cuda:0' took 1.23 ms
Module src.terrain_management.large_scale_terrain.geometry_clipmaps_warp 2e7d857 load on device 'cpu' took 2.14 ms
clipmaps updated
rock manager sampled
collider manager updated
```

Well, before run this command, `export CUDA_VISIBLE_DEVICES=0' is set (there are two RTX 3090 GPUs in the workstation) to solve the mistake as follows:
`RuntimeError: Error launching kernel '_bilinear_interpolation', trying to launch on device='cuda:1', but input array for argument 'x' is on device=cuda:0.`  

How can I solve this issue?

Thank u so much :smile:

TL-4319, 2024-10-17:
Might be silly suggestion but the scene mesh is loaded VERY high up from simulation origin. Try clicking the robot prim on the stage and hit "F". That should move your camera to the robot

dzy666fly, 2024-10-18:
> Might be silly suggestion but the scene mesh is loaded VERY high up from simulation origin. Try clicking the robot prim on the stage and hit "F". That should move your camera to the robot

:laughing:  Wonderful! It works! Maybe I need to learn how to use the Isaac Sim software in scratch. Thank u so much.

closed 2024-10-18 by dzy666fly