OmniLRS/OmniLRS #42: Unable to load robots

aryadas98, 2024-12-17:
Hi, thank you for this amazing simulator. We are facing a problem running the simulator with the default commands. The Isaac Sim window opens up, but it freezes and then the application crashes after a few minutes. I am totally new to Isaac Sim and SpaceROS but I think this is the relevant section of the console messages that would help identify the problem.

```
2024-12-17 17:49:35 [8,633ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-12-17 17:49:35 [8,633ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[8.888s] RTX ready
[9.058s] Simulation App Startup Complete
[9.070s] [ext: omni.isaac.ros2_bridge-2.5.1] startup
[9.105s] Attempting to load system rclpy
[9.130s] rclpy loaded
mesh update took 989.12 ms
2024-12-17 17:51:53 [146,808ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp -- In </Robots/husky/d455>: Could not open asset @omniverse://localhost/NVIDIA/Assets/Isaac/4.0/Isaac/Sensors/Intel/RealSense/rsd455.usd@ for reference introduced by @/home/labpc/OmniLRS/assets/USD_Assets/robots/ros2_husky_PhysX_vlp16.usd@</husky_robot/d455>. (recomposing stage on stage @anon:0x143dca40:World0.usd@ <0x143e3120>)

2024-12-17 17:51:53 [146,882ms] [Warning] [omni.timeline.plugin] Deprecated: direct use of ITimeline callbacks is deprecated. Use ITimeline::getTimeline (Python: omni.timeline.get_timeline_interface) instead.
2024-12-17 17:51:54 [147,825ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /Lunalab/Rocks/large_rocks/instancer.proto2_rock4_id0 that has not been populated
2024-12-17 17:51:55 [147,916ms] [Warning] [omni.hydra] Mesh '/Lunalab/Terrain/terrain_mesh_0' has corrupted data in primvar 'displayColor': buffer size 1 doesn't match expected size 650000 in vertex primvars
2024-12-17 17:51:55 [148,251ms] [Warning] [omni.physx.plugin] PhysicsUSD: Parse collision - triangle mesh collision (approximation None/MeshSimplification) without SDF cannot be a part of a dynamic body, falling back to convexHull approximation: /Robots/husky/front_left_wheel_link/collisions. To enable SDF collisions, sdfResolution should be > 0.
2024-12-17 17:51:55 [148,252ms] [Warning] [omni.physx.plugin] PhysicsUSD: Parse collision - triangle mesh collision (approximation None/MeshSimplification) without SDF cannot be a part of a dynamic body, falling back to convexHull approximation: /Robots/husky/front_right_wheel_link/collisions. To enable SDF collisions, sdfResolution should be > 0.
2024-12-17 17:51:55 [148,252ms] [Warning] [omni.physx.plugin] PhysicsUSD: Parse collision - triangle mesh collision (approximation None/MeshSimplification) without SDF cannot be a part of a dynamic body, falling back to convexHull approximation: /Robots/husky/rear_left_wheel_link/collisions. To enable SDF collisions, sdfResolution should be > 0.
2024-12-17 17:51:55 [148,511ms] [Warning] [omni.fabric.plugin] No source has valid data array=0x22337940 usdValid=0 cpuValid=0
2024-12-17 17:51:55 [148,511ms] [Warning] [omni.fabric.plugin] No source has valid data array=0x23d23ca8 usdValid=0 cpuValid=0
2024-12-17 17:51:55 [148,511ms] [Error] [omni.graph.core.plugin] /Robots/husky/Graphs/differential_controller/get_array_index_01: [/Robots/husky/Graphs/differential_controller] inputs:index 1 is out of range for inputs:array of size 0
2024-12-17 17:51:55 [148,511ms] [Error] [omni.graph.core.plugin] /Robots/husky/Graphs/differential_controller/get_array_index: [/Robots/husky/Graphs/differential_controller] inputs:index 0 is out of range for inputs:array of size 0
terminate called without an active exception
Fatal Python error: Aborted
```

We suspected that this is a problem with the Husky robot, so we ran the environment without any robot as mentioned in #32 and it runs fine. The Lunalab environment opens up and we can move the camera around the scene. We are facing this problem in both the native and docker installations.

TL-4319, 2025-01-03:
Hi. This seems very similar to this #25 that I encounter. What I needed to do to fix it was to install Nucleus as well. Hope that helps.

Z3ZEL, 2025-03-17:
I can confirm as well, had the same problem. A disclamer in the readme could be helpful

closed 2025-09-24 by ljburtz