OmniLRS/OmniLRS #46: /isaac-sim/python.sh: line 41:   946 Aborted                 (core dumped)

asimonov, 2025-02-12:
Hello,

I am trying to run docker version of current main branch and getting the following error:

```
root@razer:/workspace/omnilrs# /isaac-sim/python.sh run.py environment=lunaryard_20m
/workspace/omnilrs/run.py:76: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_name="config", config_path="cfg")
/isaac-sim/kit/python/lib/python3.10/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/isaac-sim/kit/python/lib/python3.10/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
{'profiles_path': 'assets/Terrains/crater_spline_profiles.pkl', 'min_xy_ratio': 0.85, 'max_xy_ratio': 1.0, 'resolution': 0.025, 'pad_size': 500, 'random_rotation': True, 'z_scale': 0.5, 'seed': 42}
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--allow-root']
Passing the following args to the base kit application:  ['environment=lunaryard_20m']
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2023.1/kit_20250212_115252.log
2025-02-12 11:52:52 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.217s] [ext: omni.kit.async_engine-0.0.0] startup
[2025-02-12 11:52:54,054][asyncio][DEBUG] - Using selector: EpollSelector
[1.958s] [ext: omni.activity.core-1.0.1] startup
[1.963s] [ext: omni.assets.plugins-0.0.0] startup
[1.967s] [ext: omni.stats-0.0.0] startup
[1.969s] [ext: omni.client-1.0.1] startup
[1.981s] [ext: omni.activity.profiler-1.0.2] startup
[1.986s] [ext: omni.gpu_foundation-0.0.0] startup
[2.013s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[2.016s] [ext: carb.windowing.plugins-1.0.0] startup
[2.043s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.183.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3080 Laptop.. | Yes: 0 |     | 8192    MB | 10de      | 0          |
|     |                                  |        |     |            | 249c      | 6f999355.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 6.8.0-52-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: AMD Ryzen 9 5900HX with Radeon Graphics | Cores: 8 | Logical: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 15388 | Free Memory: 10754
| Total Page/Swap (MB): 36239 | Free Page/Swap: 31384
|---------------------------------------------------------------------------------------------|
2025-02-12 11:52:57 [5,345ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled. Found 11 items in /sys/kernel/iommu_groups/.
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin] An input-output memory management unit (IOMMU) appears to be enabled on this system.
On bare-metal Linux systems, CUDA and the display driver do not support IOMMU-enabled PCIe peer to peer memory copy.
If you are on a bare-metal Linux system, please disable the IOMMU. Otherwise you risk image corruption and program instability.
This typically can be controlled via BIOS settings (Intel Virtualization Technology for Directed I/O (VT-d) or AMD I/O Virtualization Technology (AMD-Vi)) and kernel parameters (iommu, intel_iommu, amd_iommu).
Note that in virtual machines with GPU pass-through (vGPU) the IOMMU needs to be enabled.
Since we can not reliably detect whether this system is bare-metal or a virtual machine, we show this warning in any case when an IOMMU appears to be enabled.
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin]  
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin] -----------------------------------------------------------------------
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin] !!!!! Local system validation failed! Incorrect configuration detected.
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin]       Summary below. Details above.
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin] -----------------------------------------------------------------------
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin]  
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin]  - ECC:                         OK
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin]  - IOMMU:                       FAILED
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin]  
2025-02-12 11:52:57 [5,346ms] [Warning] [gpu.foundation.plugin] -----------------------------------------------------------------------
[5.627s] [ext: omni.kit.pipapi-0.0.0] startup

[... 74 lines cut ...]
root@razer:/workspace/omnilrs# 
```

I have the following NVIDIA card/drivers:

```
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3080 ...    Off | 00000000:01:00.0  On |                  N/A |
| N/A   47C    P5              22W /  80W |    769MiB /  8192MiB |     29%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
```

for the setup you mention I would say my laptop fits the requirements, apart from RAM size being 16gb...

any advice on what may be wrong and how to fix it?

Kind Regards,
Alexey

asimonov, 2025-02-13:
hello again.

I have just tried on workstation with 500gb RAM and A4000, but seeing the same error:

```
alexey@PSRCWS08-LINUX:~/git/OmniLRS (main)$ ./omnilrs.docker/run_docker.sh
access control disabled, clients can connect from any host
root@PSRCWS08-LINUX:/workspace/omnilrs# python.sh run.py environment=lunaryard_20m
bash: python.sh: command not found
root@PSRCWS08-LINUX:/workspace/omnilrs# cd ..
root@PSRCWS08-LINUX:/workspace# 
root@PSRCWS08-LINUX:/workspace# 
root@PSRCWS08-LINUX:/workspace# python.sh run.py environment=lunaryard_20m
bash: python.sh: command not found
root@PSRCWS08-LINUX:/workspace# cd omnilrs/
root@PSRCWS08-LINUX:/workspace/omnilrs# 
root@PSRCWS08-LINUX:/workspace/omnilrs# 
root@PSRCWS08-LINUX:/workspace/omnilrs# 
root@PSRCWS08-LINUX:/workspace/omnilrs# 
root@PSRCWS08-LINUX:/workspace/omnilrs# /isaac-sim/python.sh run.py environment=lunaryard_20m
/workspace/omnilrs/run.py:76: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_name="config", config_path="cfg")
/isaac-sim/kit/python/lib/python3.10/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/isaac-sim/kit/python/lib/python3.10/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
{'profiles_path': 'assets/Terrains/crater_spline_profiles.pkl', 'min_xy_ratio': 0.85, 'max_xy_ratio': 1.0, 'resolution': 0.025, 'pad_size': 500, 'random_rotation': True, 'z_scale': 0.5, 'seed': 42}
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--allow-root']
Passing the following args to the base kit application:  ['environment=lunaryard_20m']
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2023.1/kit_20250213_071447.log
2025-02-13 07:14:47 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.085s] [ext: omni.kit.async_engine-0.0.0] startup
[2025-02-13 07:14:48,715][asyncio][DEBUG] - Using selector: EpollSelector
[1.020s] [ext: omni.activity.core-1.0.1] startup
[1.025s] [ext: omni.assets.plugins-0.0.0] startup
[1.028s] [ext: omni.stats-0.0.0] startup
[1.030s] [ext: omni.client-1.0.1] startup
[1.051s] [ext: omni.activity.profiler-1.0.2] startup
[1.055s] [ext: omni.gpu_foundation-0.0.0] startup
[1.087s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[1.089s] [ext: carb.windowing.plugins-1.0.0] startup
[1.118s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 565.57.01     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA RTX A4000                 | Yes: 0 |     | 16622   MB | 10de      | 0          |
|     |                                  |        |     |            | 24b0      | b1e5d1e5.. |
|     |                                  |        |     |            | 2d        |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 5.19.0-46-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Xeon(R) Gold 5218R CPU @ 2.10GHz | Cores: 20 | Logical: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 514478 | Free Memory: 502556
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[2.243s] [ext: omni.kit.pipapi-0.0.0] startup
[2.248s] [ext: omni.kit.pip_archive-0.0.0] startup
[2.248s] [ext: omni.pip.compute-1.2.0] startup
[2.249s] [ext: omni.pip.cloud-1.0.1] startup
[2.265s] [ext: omni.isaac.core_archive-2.2.1] startup
[... 349 lines cut ...]
  File "/isaac-sim/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 596 in initialize_physics
[... cut ...] aces.rcl_interfaces_s__rosidl_typesupport_c, builtin_interfaces.builtin_interfaces_s__rosidl_typesupport_c, std_msgs.std_msgs_s__rosidl_typesupport_c, geometry_msgs.geometry_msgs_s__rosidl_typesupport_c (total: 140)
/isaac-sim/python.sh: line 41:    59 Aborted                 (core dumped) $python_exe "$@" $args
There was an error running python
root@PSRCWS08-LINUX:/workspace/omnilrs# 
```

it looks like it is working ok setting up the simulation, but once the omniverse starts to run it crashes?

asimonov, 2025-02-13:
just noticed that something is wrong with ros2 environment in the container:

```
root@PSRCWS08-LINUX:/workspace/omnilrs# ros2 pkg list
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.11', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/bin/ros2", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/exceptions.py", line 15, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "/opt/ros/humble/lib/python3.10/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /opt/ros/humble/lib/librcl_logging_spdlog.so: undefined symbol: _ZN6spdlog7details7log_msgC1ENS_10source_locEN3fmt2v817basic_string_viewIcEENS_5level10level_enumES6_
The C extension '/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/{distro}/Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions
root@PSRCWS08-LINUX:/workspace/omnilrs# 
```

not sure if this is relevant

asimonov, 2025-02-13:
it looks like issac uses different version of python than what ros2 installation was built against:

```
root@PSRCWS08-LINUX:/workspace/omnilrs# /isaac-sim/python.sh 
Python 3.10.13 (main, Oct 18 2023, 20:21:26) [GCC 7.3.1 20180303 (Red Hat 7.3.1-5)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
root@PSRCWS08-LINUX:/workspace/omnilrs# python3
Python 3.10.12 (main, Jan 17 2025, 14:35:34) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

https://docs.ros.org/en/humble/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system

ljburtz, 2025-09-24:
@asimonov thanks for pasting the entire log. 
The error was `ModuleNotFoundError: No module named 'em'`
this has been fixed in the latest main branch. you can pull and rebuild the docker and launch the simulation again. 

regarding ros2 within the docker. you are right that the python version has a conflict. this is fixed in the next versions of isaacsim. 
for the current version (still running on isaac 2023) the best workaround is to spin another docker with ros. the most straightfoward is to follow the instructions in https://github.com/jnskkmhr/omnilrs_ros2_demo
in particular the keyboard control works well for both husky and jackal rovers in all three environments.

ljburtz, 2026-03-03:
closing as fixed by omnilrs v2 (on isaacsim 5)

closed 2026-03-03 by ljburtz